UMSAE-Formula-Electric/SCU2.0 #16: IMU setup and integration

CPollreis, 2024-12-22:
# Overview:
Our IMU (Inertial Measurement Unit) is a sensor that measures acceleration and angular rate (yaw, pitch, and roll in XYZ direction). In the future, we aim to combine the data from this sensor and our wheel speed sensors to build a traction and launch control system. It communicates via CAN2.0B messages to our CAN bus.

# Resources:
- On Teams, all the IMU information can be found under: Software System -> Software Resources -> Datasheets -> IMU
- Take a look at the OpenIMU300RI datasheet to get a better understanding of how the sensor works
- The sensor will be integrated with our SCU look at the SCU2.0, specifically the Rx messages, to find where we will receive the CAN messages from the IMU

Evan1818, 2025-02-23:
Resolutions calculations look good according to the datasheet

IMU Part:
https://www.digikey.ca/en/products/detail/aceinna-inc/OPENIMU300RI-EVK/13147433

CPollreis, 2025-04-27:
https://github.com/UMSAE-Formula-Electric/CAN_test_messages/issues/6

^ waiting on this issue to test