UMSAE-Formula-Electric/SCU2.0 #23: Individual Sensor Testing

CPollreis, 2025-02-24:
- Make sure the watch for any hardware components you'll need for this. I checked the 3 datasheets we have on Teams and we will need this circuit:
[image]
- I'll make sure we have a sparky on this so you guys can work together

Evan1818, 2025-04-28:
[FlowmeterDocument.pdf](https://github.com/user-attachments/files/19932594/FlowmeterDocument.pdf)

Finished Individual Testing on the flowmeter. Included a document about it

CPollreis, 2025-08-29:
- tested with kenyi and angelo but it wasn't the greatest test
- measured flow rate with pump
- had an expected value with the test and they were "close enough" so they called it good
- definitely need to go back and test this when everything comes together on the car and on the test bench

- tested the flow meter indiv with the cooling loop
- thermistor was tested separately 

goal:
- test on SCU repo with all sensors
  - if possible have cooling going at the same time

closed 2025-08-29 by CPollreis